msr-consulting/exscalabar #1: Restyle sidebar

mtat76, 2015-02-27:
* All elements should be somewhat similar in size
* IP text box ought to be of size that only fits number 255.255.255.255
* Port should be the same size as the IP text box
* Connection box should be styled to be of reasonable size to fit the text appropriate to it.  Currently that text is:
 * No Connection 
 * Wait on Connect
 * Connected

 The ID of the box is ``connect`` and it is set the file ***general.js***.
* Add button for connection when connection fails at the given IP.

***These changes will be global as the sidebar is on every page.***

mtat76, 2015-06-24:
So, these have been added and are functional.  The IP and port boxes connect to the ``net`` service *and* are stored to local storage (HTML5 enabled).  Want to add the following:

1. Filter cycle parameters.
2. Calibration buttons.
3. Calibration progress indicator.

I want to restyle the indicators so they are simpler.

msrconsulting, 2015-08-10:
Moving the filter cycle and calibration stuff to their respective pages.  Will provide indicators on the page regarding filter and calibration cycles (i.e. filter position and calibration active).

lo-co, 2016-02-03:
No longer relevant to page

closed 2016-02-03 by lo-co