msr-consulting/exscalabar #7: Change axis title when changing data plotted

mtat76, 2015-04-07:
Need to figure out how to change the y-label axis name on the plot when a new plot is selected.  I have supplied a dropdown on crd.html as well as a file crd.js to put the code in...

danbek, 2015-04-19:
Don't quite understand this. Right now the plot has 5 traces, and you can select/deselect which traces are displayed by clicking on the names or circles in the legend. Changing which traces are plotting shouldn't change the label on the y-axis, right?

Do you mean that selecting different values in the drop-down should change the plot to an entirely different set of traces? E.g., one set of plots might be plotting extinction in units of inverse length, while another set would be temperatures, or pressures?

So you need to be accumulating multiple sets of data in the background, and you can choose which set via the drop-down?

mtat76, 2015-04-20:
You got it....

I should sit down with you and discuss the layout of these pages to see if you have any better ideas...

danbek, 2015-04-20:
Some time Tuesday?

mtat76, 2015-04-21:
Should be fine....when?

danbek, 2015-04-21:
My day is pretty open, say 3:00 PM?

mtat76, 2015-04-21:
Should work.  Where?

danbek, 2015-04-21:
This afternoon is looking bad now, I could stop by your place after 8:15
tonight, or we could wait for tomorrow.

Dan

mtat76, 2015-04-21:
Tomorrow is fine.

danbek, 2015-04-21:
Should we meet in Longmont right after drop-off at TPCA?

closed 2015-06-22 by mtat76