msr-consulting/exscalabar #13: Add PAS laser control functionality

mtat76, 2015-06-24:
There should be one value for each cell (5 cells total).

1. Laser modulation - this is a U8 enum with Sine (0) and Square (1).  This controls how the lasers are modulated.
2. Laser frequency - this is a float and controls the frequency at which the lasers are modulated.  This should be connected to the resonant frequency determined by the speaker calibrations.  When the system is running a speaker calibration, these values should be updating with the resonant frequency.  Otherwise, the user should be able to determine what the values are .
3. Voltage range - this is a float.  A value of 5 will produce a wave that runs over -5 to 5 + the offset.
4. Voltage offset - this is a float.  This value offsets the laser modulation waveform
5. Laser enable - boolean.  Toggles the state of the laser.  FALSE is off.

msrconsulting, 2015-08-03:
Most of this is exposed on the FP, but the functionality is not all hooked up and tested on the backend (web service VI for enabling lasers not currently available).

lo-co, 2016-06-09:
Finished with this.

closed 2016-06-09 by lo-co