msr-consulting/exscalabar #14: Style PAS plots

mtat76, 2015-06-24:
Plots need styling.  

* Plots should be contained within a single column that is centered between the controls and the sidebar.
* Plots need to resize properly.
* Add right-click functionality for 
  * Switching between different data sets
  * Autoscaling.
  * Clearing the plot data (this will have to be connected to the data service so that we can clear the data out)
* Provide easy ability to turn off different plots.

mtat76, 2015-06-29:
Autoscaling indicates that the scale will automatically adjust to the max and min.  I can't remember if most of these plots automatically autoscale, but we should have an autoscale and and an autoscale 1x option (meaning that the graph autoscales once and then sits at the new max and min until requested differently).

mtat76, 2015-06-29:
Switching between data sets means exactly what you think - f0 -> Q -> IA, etc, etc...

lo-co, 2016-06-09:
This was fixed a long time ago.  Closing

closed 2016-06-09 by lo-co