msr-consulting/exscalabar #34: Add purge switch UI element

lo-co, 2015-08-31:
Probably goes on CRD page?  See issue lo-co/exscalabar_server#11

lo-co, 2015-12-01:
Added flow and valve position controls to the CRD page.

JustinLangridge, 2016-06-08:
Testing done and identified 2 issue with these controls

1) the valve position control is not switching the valve correctly. The underlying task works fine when tested in MAX. However upon changing the valve to 'on' in the UI, it switches the valve but then immediately returns it to the off state. The 'on' light on the control remains on but this isn;t reflecting the actual valve position. 

2) The flow rate setting for the alicat isn't do anything. There are now two places where the purge gas flow rate con be set - on the CRD page and on the flow control page. The one on the flow control page works. Probably best to remove it from there and get the one on the CRD page working (or move everything to the flow control page)

JustinLangridge, 2016-06-22:
Not too far off having the functionality to align the ringdown now (and thus start testing the trace fitting etc). Any chance this one could be looked at  in next few days to allow purge gas to be used?

lo-co, 2016-06-23:
OK...so I combed the code to try to figure out why ``1`` might do what you described and the best I can come up with this is:

When the system starts, it sets the purge flow switch to ``TRUE`` (sends a digital high).  Good, this should be what you want.  However, this is not reflected in the UI because we are not polling the CVT for this value.  In fact, if you reload the page, the switch should reset to it's default of ``FALSE`` as there is no backend to the button.  I am going to fix this, so let me know tomorrow if this works.

lo-co, 2016-06-23:
Added line in ``cvt-service``:

```javascript
cvt.purge.pos = response.data.general.purge;
```

Corrected ``purge`` object in ``ex.crd.ctl.js``:

```javascript
$scope.purge = {
                pos: cvt.purge.pos,
                flow: 0.16,
                setValve: function () {
                    this.pos = !this.pos;
                    cvt.purge.setSw(this.pos);

                },
                setFlow: function () {

                }
            };
```

Also added update for ``purge.pos`` when the CVT is updated:

```javascript
$scope.purge.pos = cvt.purge.pos;
```

Flow is currently hardcoded.  Should be associated with ``AlicatI``.  And there is no body to the ``setFlow`` function.

JustinLangridge, 2016-06-23:
Just tested this and we still have the problem. I hit `Flow Ctr` to on and the valve switches but then after about a second goes back to default position.

lo-co, 2016-06-24:
OK....so tested this with Justin today and after removing the cable from the valve, we found that the signal stayed high when we 1) used MAX and 2) used the UI.  All seemed well.  Then, we plugged the cable back in and viola! Britain voted to leave the EU...I mean the valve worked.  I feel like I have seen something like this before.  Is this a grounding issue that just resolves itself?

**Anyway, the gist seems to be that the UI works.**

JustinLangridge, 2016-06-24:
HI Matt, sorry but having to re-open this issue.

testing this again today we have gone back to the not working state (same symptoms as before). To see if this is hardware or software issue I put a volt meter straight onto the digital out from the PXI. Sure enough enough we see a transient 5V when you hit then button that falls back to 0V quickly (timing of this is variable but usually within 1-2 seconds). Driven from MAX it stays on at 5V all the time.

JustinLangridge, 2016-06-27:
Matt, a bit more playing with this today and the fact that it works through MAX has got me back to convinced that this is a software rather than hardware issue. What are your thoughts on moving this forward?

lo-co, 2016-07-07:
OK...so I identified a problem with this.  We have two DIOs using the same line 

* ``General Input`` - ``Blocked Pump`` uses ``CRD_HK/port1/line3``
* ``Purge Switch`` - uses ``CRD_HK/port1/line3``

It is not clear to me how this *ever* worked.  According to the signals list, the purge switch should be on ````CRD_HK/port1/line4``.  I have switched it to this value in MAX and it should stick.  If this is not the correct placement, then switch it back and do something with the first signal.  The first signal is a 1 Hz input so it will always override any other input.

JustinLangridge, 2016-07-07:
Good catch. However hardware is configured to have the blocked pump on p1.4 and purge valve on p1.3.

Have switched these in MAX and confirmed that purge valve is now working correctly via the UI. Closing issue.

closed 2016-07-07 by JustinLangridge