msr-consulting/exscalabar #51: Add Meerstetter TEC controls

lo-co, 2016-02-03:
These are completely missing.

JustinLangridge, 2016-06-10:
Yes please, ir would be good to be able to start testing the PAS cell temperature setpoints (we've not looked at this at all yet)

JustinLangridge, 2016-06-23:
Dave has stated testing the Meerstetters today by manually sending commands. Note that all PAS TECs seem to be working, but we have to fix a connector issue for the humidifiers meerstetters. This should be quick.

Once we have UI controls we can start testing this more thoroughly

lo-co, 2016-06-30:
Control for the setpoint is added to the UI along with PID and temperature/power selection.  Currently

* there is no backend for the PID.  I am not going to add this for this phase as I don't think it is critical
* the temperature/power selection is not working.  Will work to get this going, but probably not critical also

Currently, the way the UI components for the TECs works is 

1. when the server is up and running, the ``cvt`` service will poll the json string returned representing the CVT on the server side
2. It will look under the heading ``device`` for devices with ``type`` ``mTEC``.
3. The client will then add elements to the UI using ``ng-repeat`` with current values found in the CVT.

The UI elements **will not** be displayed *until* the first successful CVT return from the server.

JustinLangridge, 2016-06-30:
Hi Matt, can confirm all UI setpoint controls work. 

Note that the table also populates with controls for the 2 humidifier TECs. This isn't desirable as they are controlled on the humidifiers page. Can these be changed?

lo-co, 2016-06-30:
I will look into this.  May have to create a key that says they are reserved...

datid, 2017-02-06:
The Meerstetters now have set point controls and output controls ( to turn their out put on and off ) which work.  There is no need for PID changes, and the Humidifier page is populated - mostly working, but plots could do with some finessing.

closed 2017-02-06 by datid