msr-consulting/exscalabar #61: Add TE Tech visualization and control

lo-co, 2016-03-12:
Add plots and controls for the TE Technology TEC.  The TEC will output power and two temperatures associated with two different thermistors.

For inputs, we will need access to the PID parameters, temperature setpoint and the heating and cooling side factors (``htx`` and ``clx`` in the configuration file).

lo-co, 2016-03-13:
The visualization is in there.  Not sure if I like how it is - everything on one plot.  Directive is ``ExTetechplotDir``.

lo-co, 2016-03-13:
Added controller ``ExTetechCtl`` that contains ``$scope`` variables for:

* ``P``
* ``I``
* ``D``
* ``clx``
* ``htx``

JustinLangridge, 2016-06-27:
Matt, Dave can elaborate here but basically he has found today that the commands from the TE tech controls on the UI are not making over to the server. He can send the same commands by explicitly entering them in the browser and change the temp setpoint and P, I, D variables successfully.

lo-co, 2016-06-28:
Yes...I broke it.  On purpose.  Trying to simplify and reuse some stuff.  One thing that will need testing.

JustinLangridge, 2016-06-30:
Should this be fixed now? Tried it today and still not communicating via UI

lo-co, 2016-06-30:
Meh...wasn't guaranteed.  I need to see what is going on with this.  Maybe tomorrow morning if I can get up early enough?

lo-co, 2016-06-30:
Were you able to see where the plots were supposed to be?

lo-co, 2016-08-02:
Is this working?

JustinLangridge, 2016-08-03:
The plots are coming through and we can set the setpoint. The P, I, D, htx ad clx controls don;t appear to working. However, I'd be happy with removing these from the UI and having them set only in the ini file if this is doable?

JustinLangridge, 2017-02-08:
Removed extra controls, happy to close

closed 2017-02-08 by JustinLangridge