msr-consulting/exscalabar #65: Ability to stop/start waveform streams

lo-co, 2016-03-14:
Add ability to stop and start the waveform streams in the JSON data returned from the server.

JustinLangridge, 2016-06-08:
A comment here that the system seems on the limit with the waveform data. When the user uses these buttons to start/stop the PAS or CRD waveform streams (or even changes the plot variables) it often freezes the system for 10s seconds

lo-co, 2016-06-09:
I have not seen this, but it should not be the case so long as you have 100 Mbit routing?

JustinLangridge, 2016-06-10:
Came up against the same issue when testing the photodiode signals in the UI today.

MAX was having no problem seeing the sinusoidal signals from the PDs but the UI plots weren't showing anything. We'd get occasional blips but it looked like they couldn;t keep up

JustinLangridge, 2016-06-27:
Can almost close this one but one remaining question. It appears that the send waveform button for the PAS  stops PAS waveform data coming across in the CVT,. in contrast when you stop the CRD waveforms via the UI button, the UI plots stops but the PMT waveform data still comes across in the CVT.

Is this a correct behaviour?

JustinLangridge, 2017-02-08:
This still needs to be looked at