msr-consulting/exscalabar #69: Interlock message is not responding appropriately

lo-co, 2016-03-18:
Always says lasers are contained.

JustinLangridge, 2016-05-18:
In addition part of lthe laser safety case is that the lasers will be powered 1 by 1 when the lid is off. INd addition to updating the interlock message, can the software set either all laser powers to 0% or turn off  all laser enable signals (if we get them working) when it senses the interlock has been triggered

lo-co, 2016-05-19:
Hmmmm...is this tested in MAX?   I will take a look at.

The Interlock is monitored in the ``Controller``.  We will send a message if we see the interlock is active to shutdown the lasers.

lo-co, 2016-08-02:
Updated sidebar controller so now it indicates what is actually going on with the interconnect.  Now need to provide proper indication of when a laser is on.  Laser is on when:

* Interconnect is ``TRUE``
* *One* of the lasers is powered.