msr-consulting/exscalabar #77: UI control scales

JustinLangridge, 2016-05-18:
Currently most UI controls for analog output signals are in volts. Can these be changed to the following (with units indicated in UI):

CRDS laser k: currently 0-10V output. Change to full scale 0-5V output and display as 0-100% in UI (linear scaling with 5V = 100% power)

PMT gain: currently 0-10V output. This output is fine but display as 0-100% scale (linear scaling with 100% = 10V)

PAS lasers:combine Vrange and Voffest to a 0-100% laser power scale, with Voffest always set to half of Vrange max.

lo-co, 2016-05-30:
Have now done this for the PAS.  Having some validation issues (range is not set correctly).  Will fix this.  The range is set via the variable ``maxRange`` in ``exPaslasCtl``.  We need to update the server so that you can only specify voltage range there and we will halve the range.

lo-co, 2016-05-30:
Ultimately, would it be more pleasing to have these as sliders?

lo-co, 2016-05-30:
So, this is now updated for the CRD gains also.  The PMT gains are an array, so you have to map the array to multiply a constant:  It looks something like

```javascript
$scope.pmt = cvt.crd.kpmt.map(function(x){return x/maxPMTGain*100;});
```

Solutions to this problem are given in the SO post [here](https://stackoverflow.com/questions/8454977/how-do-i-multiply-each-member-of-an-array-by-a-scalar-in-javascript).

The max gains are given by the variables ``maxPMTGain`` (set by default to 10) and ``maxLaserGain`` (set by default to 5).  The file of interest is ``js/ex.crd.ct.js``.

commit a39ef7f7 (2016-05-30): Closes #77.
closed 2016-05-30 by lo-co