msr-consulting/exscalabar #80: Control units

JustinLangridge, 2016-05-18:
General request to add units to control labels in the UI (and if possible limit user inputs to a sensible values)

lo-co, 2016-06-09:
Need an example of this.  Most controls have labels or they have tips that explain exactly what you are entering.

JustinLangridge, 2016-06-10:
Ok can see the tips now, these help a lot. Here are the outstanding uncertainties:

**System controls/CRDS page**

- purge Q MFC needs units
- DC on laser. Is this actually percent (i.e. 0-100) as the tip suggests or does it want a value between 0-1

**System controls/PAS page**

- Speaker Vrange and Voffset: what is the acceptable range and what are the optimal values?
- Laser setting. Please could we change this label to 'laser power' instead of 'laser setting'

**System controls/flow control page**

- units for flow set points

**System controls/common page**

- I don't know what the tr setting is for?

**Housekeeping**

- units for the TE Technology controller (we still need to test these controls too!)