msr-consulting/exscalabar #88: CRD laser gain control

JustinLangridge, 2016-06-14:
On the 6704 card we have 3 laser gain signals (AO0-2) to control the power of the 3 CRD lasers.

Currently AO0 and AO1 are connected to the UI controls (labelled red laser k and blue laser k). However we have no control of AO2 and thus can not change the power of this lase. Worse, The output is permanently 0V so the laser won;t power up.

Can we add a control for the power of this laser n the UI (and at the same time add a power button linked to its laser enable signal so we can turn it on and off - P1.7 on CRD_HK

lo-co, 2016-06-17:
Odd...enable functionality isn't even in the UI right now.  Code looks like:

```javascript

            $scope.setEnable = function (index) {

                $scope.laser_ctl[index].en = !$scope.laser_ctl[index].en;
                var enabled = $scope.laser_ctl[index].en;
                switch (index) {
                    case 0:
                        cvt.crd.eblue0 = enabled;
                        break;
                    case 1:
                        cvt.crd.eblue1 = enabled;
                        break;
                    case 2:
                        cvt.crd.ered = enabled;
                        break;
                    default:

                }

            };
```

No http backend?

lo-co, 2016-06-17:
``Change Laser Gain MSG::Do`` method improperly referencing the CVT.  In the CVT, we have ``kblue0``, ``kblue1`` and ``kred``.  We will replace this with an array with three elements that are:

0. Red
1. Blue0
2. Blue1

Need to remove the reference to the original elements and make sure these are being broadcast properly.

lo-co, 2016-06-17:
Fixed CVT init issues in ``Controller::Initialize CVT``.

lo-co, 2016-06-17:
CRD CVT side is a disaster.  Added elements to ini file:

* ``enblue0``
* ``enblue1``
* ``enred``

These are booleans and represent the enable state of the lasers.  There is now a new CVT entry under ``crd`` called ``enabled`` which represents these values (in the order red, blue0, blue1).  

There ``Change Laser Enable State`` message object now has three entries representing the three values (pretty certain this was NEVER called before).  Also, changed the ``Do`` method for this message to call the corresponding ``Command`` object ``Laser Enable CMD`` for consistency.  As such, removed all code handling DIO to the CMD and no log message emits at this level - shifted to command.  Finally, the CVT is updated above if the CMD is successfully received.

lo-co, 2016-06-17:
Need to do the same as above for the `Change Laser Gain MSG``.  Going to add also an issue that must be handled - CRD initialization does not seem to take into account the INI file.

lo-co, 2016-06-20:
Tested UI - the enables are making there way to the server (not sure if the server is good yet). Made some adjustments to several of the client files to make this happen...  Not checked in yet.

lo-co, 2016-06-20:
CVT not connected properly with gains - page reloads result in gains reseting to 20%.

commit 58aff36c (2016-06-20): Fixed laser gain return on CVT.  The value is stored now down on the server.  This closes #88.  **This does not mean that the issue is resolved.  Need to verify on the server side that things are changing as they should.
closed 2016-06-20 by lo-co