msr-consulting/exscalabar #93: CRD digital modulation controls not working

JustinLangridge, 2016-06-22:
The controls for setting the CRD laser frequencies used to work but have problems now that the extra blue laser control has been added.

For example I change the frequecny on blue laser 1 control and the frequency of the red laser also gets changed?

Can we have only one set of frequency and duty cycle controls for the blue laser modulation (Ctr0) and one for the red laser. We only need 2 sets of blue controls for the laser enable signals and the analog modulation outputs (k values)

In addition can you check the routing from these controls are going to the right counter outputs - this isn't right currently

lo-co, 2016-06-22:
So, changing the ``Blue Laser 0`` frequency and DC should change the ``Blue Laser 1`` frequency and DC and vice-versa.  From a coding standpoint, this is the best way to do this.  Ultimately, we would like these two independent, so I am going to keep this the same for right now.

When you change ``Blue Laser 0`` settings, does this change the ``Red Laser`` settings?  Also, if you change the ``Red Laser`` settings, does this change either of the blue settings?

Regarding routing, I will check this out.  There are several places where the command can get jumbled.

JustinLangridge, 2016-06-22:
Changing the red laser settings does't change the blue settings. I can't remember for `Blue Laser 0`impact on `Red Laser`

If you are on the UI you can see this behaviour though just by entering numbers into each of the frequency controls.

JustinLangridge, 2016-06-24:
OK so good news is that we now have the crd laser digital modulation working for the first time. Still some issues with the 3 CRD laser controls though. Here is what they are currently doing:

**Red laser**

`frequency` working (albeit with initialisation problem noted elsewhere)
`DC` not working but we already have this raised in another issue
`k` not having any impact of the voltage coming out of the PXI, which is fixed at 1V

**Blue laser 0** (please can we rename to Blue laser lower)

- frequency coming out of this channel is being controlled by the blue laser 1 control. The Blue laser output 0 frequency control has no impact
- k control does nothing

**Blue laser 1** (please rename to blue laser upper)

Same issues as red laser (i.e. k no longer working)

JustinLangridge, 2016-07-27:
Closing this issue as no longer relevant now that we have changed the modulation setup

closed 2016-07-27 by JustinLangridge