msr-consulting/exscalabar #94: Waveform update rate is too slow

lo-co, 2016-06-22:
The CRD waveform update rate is too slow and we will not be able to calibrate using the current rate.

lo-co, 2016-06-22:
Not sure what is going on here.  Sounds like a bandwidth issue but we should have plenty of bandwidth to handle this.  We should look at hardware settings to make sure we are good and remove bottlenecks in the network.

JustinLangridge, 2016-06-22:
I'm pretty sure Dave has already looked this from a hardware perspective with no joy. Will let him add details if there are any to report

JustinLangridge, 2016-06-23:
Dave and I have looked at this again this morning. Several observations:

- with all waveform data being sent network usage is about 6 Mbps - shoudl be fine 
- turning off the PAS waveforms has a big effect. Network usage drops to 370kbps. However CRD waveform plot update rate is still slow
- turning off the crd waveform plots seems to stop the UI plotting but the data is still being sent across

Although our laptop port will only do 100Mbps I think these tests suggest we don't have an issue with the netwrk capacity. is there some aspect fo the plotting that could be slowing things down?
we have a problem somewhere

lo-co, 2016-06-24:
So, here are two snapshots of network usage.  In the first snapshot below, it takes about 1.25 s for 652 kb of data to transfer.

[image]

In the one below, it takes about 200 ms for the same amount of data to transfer.  What is going on?

[image]

lo-co, 2016-06-24:
Here is a longer timeline.  From this, you can see that the data regularly takes ~250 ms to return, then we have these periods where it takes ~1.25 s to return and the rare time where it takes > 3 s to return.

[image]

lo-co, 2016-06-24:
Saved a capture via wireshark.  You can see these packets getting all chopped up when things go sideways.  Still not certain why.  Probably need to talk to folks at NI.

lo-co, 2016-06-24:
To be clear - this has nothing to do with graphing.  These logs were captured when no graphs were visible.  I am worried we have a server problem.

commit 5cd8fc90 (2016-07-28): Closes #109, closes #107, closes #111, closes #99 and closes #94.
closed 2016-07-28 by lo-co