msr-consulting/exscalabar #97: PAS waveform plot order jumbled

JustinLangridge, 2016-06-30:
The PAS cell labels in the PAS waveform data plot and the table at the bottom of the page are jumbled. 

Here is what they should be:

**Current UI label -  actual cell**
cell 5 - cell 4
cell 4 - cell 3
cell 3 - cell 2
cell 2 - cell 1
test 1 - cell 5

lo-co, 2016-07-01:
Justin, I have traced the data from the web service to the UI and I am unable to see what you are seeing here.  Let's take a look at this together, maybe tomorrow.

JustinLangridge, 2016-07-01:
Hi Matt, ok lets take a look. The ordering seems to change each time the system is restarted. Sometimes it is right but often not. 

Where is this ordering actually defined?

lo-co, 2016-07-01:
By restart, do you mean restart the client or the server?

lo-co, 2016-07-02:
Justin, I looked at this today after you said they were jumbled.  Start at ``ePAS::Get Data`` where we retrieve the waveforms, the order was consistent to the UI.  Maybe we can go over this Monday and you can tell me what you think is going on because I have yet to see it.

JustinLangridge, 2016-07-03:
Hi Matt, ok let's look tomorrow. The CRDS waveform plot seems to have the same issue so we could look at that at the same time.

lo-co, 2016-07-04:
OK...so this seems to be an observable problem likely do to the DMA FIFO that transfers data between the FPGA and the PXI.  The cells are always in order, it just seems to be that the start is not where it should be.  Also, we should always have multiples of 5 in the buffer but it would seem we are catching the buffer at odd times and getting instances where we don't have multiples.  Not sure if this is an issue yet...

datid, 2016-07-04:
Seems like forcing a reset on the FPGA when it is configured has sorted the issue.  I think that if the real time code is stopped, the FPGA keeps going, overflows the FIFO, and so when the realtime is restarted it's any ones guess what is in the FIFO.

closed 2016-07-04 by datid

lo-co, 2016-07-06:
This should not be the case.  When we shutdown the RT server, we also call a close on the FPGA server.  That being said, this may be consistent with the observation that the FPGA throws a warning concerning initialization some times.  Probably need to probe this harder, but going to leave this closed so long as you all are happy.