msr-consulting/exscalabar #98: PAS speaker setup questions

JustinLangridge, 2016-06-30:
Answers to these questions would be very helpful so that we can set up the PAS speaker/mic hardware properly:

- What do speaker Vrange and Voffset actually set? As far as I can tell Voffset does nothing. 

- What do you set these values in AOP

- I have set the hardware gains in the 5 PAS cell top boxes to all be the same. Note that the amplitudes of the microphone returns when on speaker are not the same - this is odd. Do you see this in AOP?

- When you look at the time domain mic signal with speaker on it appears to change from second to second. I don't understand why this isn't constant.

lo-co, 2016-06-30:
> What do speaker Vrange and Voffset actually set? As far as I can tell Voffset does nothing.

Just what they describe.  The reason I think you see nothing is that we take the DC output and AC couple it before sending to the speaker.  The DC offset is required by the speaker due to some issues that are noted when you are flipping bits between +0 and -0; this is when the MSB switches between 1 and 0 causing more significant glitches.  See [here](http://digital.ni.com/public.nsf/allkb/9B13E25B0B0EC197862562B4007D2E33) for some discussion (I can't find the paper that recommended what we do, but there is one with respect to the R-series card).

> What do you set these values in AOP

Not sure, but I would start small and go until you start to saturate the signal (around 32k).  Going beyond this could 1) damage the speaker and 2) distort the spectrum returned.  We have recently been discussing this with regards to the AOP.  You can actually tune the speaker **and** mic gains on Dan's board.  You will want to tune these so that you get maximum SNR while also being able to get a good speaker signal.  Turning the mic up to much will potentially 1) increase the noise floor and 2) force you to decrease the speaker output signal to the extent that you lose important features.

> I have set the hardware gains in the 5 PAS cell top boxes to all be the same. Note that the amplitudes of the microphone returns when on speaker are not the same - this is odd. Do you see this in AOP?

Yes - see above about mic gains.  

> When you look at the time domain mic signal with speaker on it appears to change from second to second. I don't understand why this isn't constant.

It shouldn't - I will take a look at this.

JustinLangridge, 2016-07-04:
Verified today that apparent change in PAS signal s due to decimation of data. Actual waveform is not changing.

closed 2016-07-04 by JustinLangridge