msr-consulting/exscalabar #99: Scaling on time domain plots

JustinLangridge, 2016-06-30:
Can we add functionality to the time domain plots to stop them from autoscaling

lo-co, 2016-06-30:
Should be there - ``Autoscale 1x`` will force an autoscale once at the current range.  More than this we are going to likely have to hold off for a bit as it is not a huge implementation, but it will require a little work.  What say you?

lo-co, 2016-07-02:
So....Justin.  I think this is in the documentation for the UI (but maybe not easy to find).  You can use the mouse to change the scales.  ``left-click-drag`` down zooms the y-axis; ``left-click-drag`` right or left should zoom the x-axis.  To get back to the original settings, double ``left-click``.  

I will make sure the functionality above is there for all plots

commit 5cd8fc90 (2016-07-28): Closes #109, closes #107, closes #111, closes #99 and closes #94.
closed 2016-07-28 by lo-co