msr-consulting/exscalabar #102: Big delay on waveform plots

JustinLangridge, 2016-07-04:
Have noticed that the waveform plots are running around 4-5 seconds behind actual events. I.e. I change the laser power on a CRDS channel and the PMT shows the change 4-5 seconds later.

This is a different issue to #94 as it happens even when plots are updating regularly each second.  

Can this delay time be shortened. Optical cell alignment will be hard with this delay

JustinLangridge, 2016-07-18:
Can we bump this up the priority list please Matt. As we currently stand we can't use the UI for alignments as the delay time is too long. 

Dave has had a look and thinks the issue might be introduced at startup where we start to put data in buffers before it is read out and thus introduce a lag. Help needed to fix this though.

lo-co, 2016-07-18:
Data is not buffered so this is not the issue.

lo-co, 2016-07-19:
So, this is actually a server issue.  The jiggering of all of the PMT outputs has really kind of hosed things.  See lo-co/exscalabar_server#145.

closed 2016-07-19 by lo-co

lo-co, 2016-07-25:
I can't actually see an issue with this.  I can match times with the stamp that is part of the CRD data structure and that on the front panel.  The client side does not buffer.  I think this is something that we are going to have to look at closely together tomorrow.