msr-consulting/exscalabar #107: Change CVT/Data Retrieval Rate

lo-co, 2016-07-26:
The fixed 1 second is making things more difficult, especially for cell alignment.  The server will have to be synced so that when the data string is the same, the json object will be empty (similar to what we do with the cvt).

lo-co, 2016-07-27:
Changed this to 100 ms and all seems ok.  

This checks to see if the data has changed.  If it hasn't, then the status code is set to ``204``.  If the data ID array is empty then the status code is set to ``209``.

JustinLangridge, 2016-07-27:
Not sure if this is the cause but the system is now having issues with update rates. The UI clock is not ticking along each second, but stops and starts, often jumping on several seconds. 

Several system restarts have not helped.

JustinLangridge, 2016-07-27:
Dave has changed this back to 1000ms which has reverted back to normal behaviour. Needs looking at with respect to why it wasn't working with 100ms

commit 5cd8fc90 (2016-07-28): Closes #109, closes #107, closes #111, closes #99 and closes #94.
closed 2016-07-28 by lo-co