msr-consulting/exscalabar #118: CRDS Triggering

mcotterell, 2016-08-10:
The CRDS waveforms do not appear to be triggering correctly in UI, but still recording correct tau values. Interestingly, triggering seemed Okay when frequency was reduced to 1000 Hz from 1500 Hz, but issue there when frequency increased back to 1500 Hz. A system restart seemed to help, but the issue re-emerged later on in the day.

mcotterell, 2016-08-10:
Perhaps I should have put ticketed this as a server issue?

lo-co, 2016-08-10:
Probably.  For right now it is best not to play with the frequency.  The frequency should sit at 2 kHz and remain.  We have been successful with this over the past couple of weeks so I suspect this is an issue with changing the frequency.

There are two issues (lower priority) that should be raised in relation to this:

* provide flexible duty cycle
* robust response to changing frequency

lo-co, 2017-02-08:
This looks like this is directly related to exscalaber_server#159.  This should probably be closed and fixed on the hardware side.

closed 2017-06-29 by lo-co