msr-consulting/exscalabar #119: UI ozone lamp power control

JustinLangridge, 2016-09-21:
Can we hook up this control please?

lo-co, 2016-09-21:
What does this refer to?  All of the controls on the ``Common`` tab are currently hooked up.  The power array is sent via http as an unsigned byte which is converted to an array of bools which is then fed to the ``DAQmx Write`` VI.  Are you referring to something else?

lo-co, 2016-09-21:
So, I tested this functionality and the message get to where it is supposed to be.  I think that you might be referring to something else.  I will dig around...

lo-co, 2016-09-21:
OK...never mind.  I see what you want hooked up - ``O3 Lamp`` on the calibrations page.

lo-co, 2016-09-21:
Added CVT entry in the ``calibration`` section called ``lamp_ratge``.

lo-co, 2016-10-04:
Added entry in message ``Calibration::Change Frequency MSG::Do`` to change the CVT value.

We are going to make the value that is entered a rate rather than a percentage simply because I am not certain what the top rate is right now (no good documentation at present.

JustinLangridge, 2017-02-08:
Currently we have an indicator for the lamp frequency on the UI cal page. it would be good to have a control here so that we can chnage lamp power when not operatign in an ozone auto-cycle

lo-co, 2017-02-08:
Indicator only?  That shouldn't be.

commit 1385d44e (2017-02-13): Closes #119
closed 2017-02-13 by lo-co