msr-consulting/exscalabar #120: Integrate web based UI into server

lo-co, 2016-10-18:
The requirements are that this must be light-weight (< 1 MB I believe) and that all files are properly referenced.  Already found a difference in the URL naming (using LiveReload in gulp produces a url without the ``index.html``).

lo-co, 2016-10-18:
After some jiggering of the code, this is up and running and all looks pretty good.  You can catch this on ``192.168.101.214:8001/xService/index.html``.  With this addition, the user will no longer be required to run ``gulp`` unless they want to make changes to the html or js.  

Need to make this a little more user friendly with gulp putting files into the proper folders...

lo-co, 2016-10-19:
Not fully tested but appears to be working now.

JustinLangridge, 2017-02-08:
Done. Closing

closed 2017-02-08 by JustinLangridge