msr-consulting/exscalabar #125: Complete Calibration page

lo-co, 2016-12-15:
not complete

lo-co, 2016-12-15:
So, with the most recent check in we have:

* ability to remove steps
* popup dialog indicating to the user  that they have not saved the most current version of the calibration
* autopopulation of calibration with default calibration

cathrynjfox, 2016-12-19:
Tried running this today (19th Dec). Problems include:

- The settings for the flows on the O3 and O2 flow controllers don't seem to work

- Not sure that setting the ozone lamp frequency (or % as is currently labelled) actually does anything

- After clicking 'start cal' the labview block diagram pops up flashing - after clicking continue a few times it then runs.

- After running the cal for about 2 minutes, everything (lasers, pump, TECs) suddenly turns off for no discernible reason.

lo-co, 2016-12-19:
This comment is on the wrong page - you don't have the most recent version of the server.  We need to pull that.  Sorry about the confusion.

JustinLangridge, 2017-02-08:
This is generally working. Need to look at changing some of the labels (e.g. O3 lamp is a frequency not %). Will the added functionality indicated in the initial issue (e.g. for file selection) still be implemented?

lo-co, 2017-04-05:
Changed O3 lamp to frequency in steps (found in ``ex.calibration.ctl.js``).

I will add a drop down - this will take a little more work (one extra step I believe and some server-side changes).

lo-co, 2017-04-05:
Added a dropdown that provides two options - ``ozone`` and ``pressure``; ``ozone`` is default.  The page before this change showed what would be displayed if ``ozone`` is selected.  If ``pressure`` is selected, the selections are essentially truncated such that there are no ozone related steps.  

Remaining TODO:

* Provide two calibration files on server side - one for pressure and one for ozone.
* Allow for selection on server side for saving *and* running purposes.
* Make sure that when the selection is changed, the correct file is retrieved with the steps pertaining to that cal in the right hand box on the UI.

lo-co, 2017-04-05:
Associated with commit 6389105.