msr-consulting/exscalabar #128: Device status page

JustinLangridge, 2017-03-17:
For the future: add a page of indicators to show which devices defined in ini file (or known to be part of standard system config) are connected and communicating successfully. This would enable the user to quickly verify that all expected devices are  present.

lo-co, 2017-04-05:
This is now there on the configuration page.  The devices will show up grouped based on device type and will show the respective ID and the label as well as status.  If the device is not connected, the label will be 'no-op'.  Each device has an indicator - red == not connected while green == connected.

lo-co, 2017-04-05:
I unfortunately thought I was working on #124 but was off by 4 so you got a bonus...

closed 2017-04-05 by lo-co

JustinLangridge, 2017-04-05:
Matt this functionality looks really good. Only issue is that the status of each device that appears first in its list is always shown as inactive. See picture below where all of the equipment is switched on. The devices shown as inactive are showing data on the other UI pages.

[image1.JPG]

Closed #128<https://github.com/msr-consulting/exscalabar/issues/128> via 79fc23e<https://github.com/msr-consulting/exscalabar/commit/79fc23ed8a3c5026b62a1e25f5ab5cfe036dbe12>.

datid, 2017-04-05:
I trimmed the whitespace off the device names - now it is all green...

lo-co, 2017-04-06:
Hmm..no image.  But, I guess that is why the first Vaisala was always off (just assumed it was and didn't check it on the page).  Nice catch, Dave.

commit 79fc23ed (2017-04-05): Fixes #128.